FreshTomato-Project/freshtomato-mips #4: Netgear N600 WNDR3400v3 - blue wifi light is off if 2.4GHz wifi is enabled while 5GHz wifi is disabled

NintendoManiac64, 2025-02-02:
I just successfully installed the following firmware yesterday:

- https://freshtomato.org/downloads/freshtomato-mips/2024/2024.5/K26RT-N/freshtomato-WNDR3400v2-K26MIPSR2_RTN-USB-NVRAM64K-2024.5-VPN.zip

The very first thing I did was clear the NVRAM.

The afterwards, when I disable the 5GHz wifi but keep the 2.4GHz wifi enabled, the router's blue wifi light gets turned off. If I reboot the router, the blue wifi light still remains off.

I don't know if this is intentional, but that blue LED seems to be purely tied to the state of the 5GHz wifi exclusively.

txnative777, 2025-02-14:
I don't know if this is intentional, but that blue LED seems to be purely tied to the state of the 5GHz wifi exclusively.

That is correct.

NintendoManiac64, 2025-02-14:
> That is correct.

It's correct that it's intentional? Or it's correct that it is tied exclusively to the state of the 5GHz wifi?

Because if it's intentional by Tomato rather than Broadcom, then personally I find this sub-par behavior as I'm using it with device(s) that don't even support 5GHz wifi (and even if it did, I live in the boondocks where 2.4GHz congestion isn't an issue and my internet is only ~15mbps down, not a typo). Heck one use-case is even with a NIntendo DS (not even DSi) which is limited to wireless B and WEP security.

txnative777, 2025-02-15:
FreshTomato only nothing to do with Broadcom, the led's can be controlled by user intervention if needed.

I understand your needs and that is fine the WNDR3400v3 should be able to handle them as well and if you have other questions you could go to the linksysinfo.org tomato forum to post help there instead of here.

Regards