FreshTomato-Project/freshtomato-mips #5: upgrade issues on MIPS router using wl high

txnative777, 2025-02-22:
mentioned [@linksysinfo forum](https://www.linksysinfo.org/index.php?threads/noticing-issue-with-rt-n-usb-mips-admin-upgrade.78662/)

I cleaned my sources from the pull related to commit:2792720dc2d8faf8cc51b1dc9c59f80b011cfa04.

Again no change, either I would receive the "Error reading file" clicking back and I am back at the upgrade.asp with the file still in place.

Trying again, the timer will count down to 2:24 or so seconds then the page goes blank. During both tries the firmware did flash but failed to goto the next screen of counting down page of successful flash.

For now, I only tried this attempt on the E3200 VPN image with the clear nvram during the upgrade the next time I will leave the clear nvram blank and see if the upgrade goes without a hitch.

Regards

txnative777, 2025-02-22:
Still the same results on upgrading without clearing nvram, the upgrade page does not transition and fails once the counter during the upgrade went to 3:19 seconds the page is fails.

Refreshing the page will come back to "Error reading file" clicking back take me back to the upgrade.asp with the image selected.

pedro0311, 2025-02-24:
@txnative777 : I'll prepare for you a special image, just tell for what router.
It has to be with USB, and you need to change logging to USB - logging will be extended and maybe it will explain something to us here.

txnative777, 2025-02-25:
The Linksys E3200.

It maybe just these models, I don't have a problem with the F7D4302 model.

Regards

txnative777, 2025-02-25:
Just to add to this issue with upgrade, I have been having issues with cgi errors after I setup https. Once I had set my user and password and **save**, the page does not transtion to the https://192.xxx.xxx.xxx/admin-access.asp.

> Feb 25 12:46:29 unknown daemon.warn httpd[2980]: invalid ID '' from 192.168.5.147 for /tomato.cgi

I manually set https://followed by the ipaddr/

Using google chrome, debian 12 all updated, firefox experience displays same issue.

Looks to be isolated to these models?

txnative777, 2025-02-25:
Just a quick update on the upgrade issue.

So far I tested a theory by disabling Wifi using wps button so far on my Belkin N600 DB Wireless N+ and Netgear WNDR3400v2 and both times the upgrade went through as designed.

The second mentioning from the cgi issue with HTTPS, now before I enabled HTTPS I simply pressed save and watched the normal screen transition, spinning wheel, seconds count down then back to admin-access.

Next I enabled HTTPS once again except added my user/password, then pressed save and again worked as entended 

TBH, I don't feel this is a browser issue just only related to these models only however I will check again to verfiy for the cgi issue as well the upgrade.

Now if it does or is confirmed, do you feel you have a fix? If not I can at least post some of these findings in the forum that seems to be related only to this partitcular supported models so that these issues can be placed on a low priority until a fix may come later, I know you're very busy I wouldn't want to keep away from other updates to the sources.

I will post later.

txnative777, 2025-02-26:
After repeating the steps mentioned do appear to allow the upgrade to initialize, clear nvram, reboot and the cgi problem appeared to be work properly.

Regards

pedro0311, 2025-02-27:
> Looks to be isolated to these models?

@txnative777 : Do you mean just the E3200 or others as well?

pedro0311, 2025-02-27:
https://bitbucket.org/pedro311/freshtomato-mips/downloads/

Upgrade to 001 first (disabling WiFi), and then - try to upgrade normally to 002.

txnative777, 2025-02-28:
@pedro0311 yes, I only have the three E3200, F9K1102v3 and WNDR3400v2 and they would not start or finish the upgrade, with or without clearing nvram selected.

The cgi issue would only work if I pressed saved before, then after I could enable "HTTPS" setup user/password and save with not issues.

I can do these test builds and get a **yes** or **no**.

txnative777, 2025-02-28:
001-Test flashed properly with wifi disabled using wps button, 002-Test stopped after 1:24 into the upgrade. On 001-Test the upgrade picked up around 1:00 then flashed and rebooted clearing nvram.

The HTTPS cgi issue still persists without pressing save first before setting.

pedro0311, 2025-03-03:
https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-E3200-K26MIPSR2_RTN-USB-NVRAM60K-2025.1.003-beta-test_upgrade-VPN.bin

Please upgrade first to 003 with wifi disabled, and then again to 003 with wifi enabled.
Let's focus now only on upgrade issue.
Also check your settings in Admin -> Debugging: "Enable segfault logging" and "Kernel printk log level" -> Debug.

pedro0311, 2025-03-03:
Oh, and paste here the syslog from the 2nd upgrade, form text: "STARTING UPGRADE".

pedro0311, 2025-03-03:
One more: are you upgrading by cable?

txnative777, 2025-03-03:
One more: are you upgrading by cable?

Yes I am.

STARTING-UPGRADE:
Jan  1 01:00:09 unknown syslog.info syslogd started: BusyBox v1.37.0
Jan  1 01:00:09 unknown user.notice kernel: klogd started: BusyBox v1.37.0 (2025-02-28 14:29:07 CET)
Jan  1 01:00:09 unknown user.notice kernel: Linux version 2.6.22.19 (root@tomato) (gcc version 4.2.4) #1 Mon Mar 3 13:35:03 CET 2025
Jan  1 01:00:09 unknown user.warn kernel: CPU revision is: 00019749
Jan  1 01:00:09 unknown user.warn kernel: wombo_reset set to gpio 23
Jan  1 01:00:09 unknown user.warn kernel: Determined physical RAM map:
Jan  1 01:00:09 unknown user.warn kernel:  memory: 04000000 @ 00000000 (usable)
Jan  1 01:00:09 unknown user.debug kernel: Entering add_active_range(0, 0, 16384) 0 entries of 256 used
Jan  1 01:00:09 unknown user.info kernel: Zone PFN ranges:
Jan  1 01:00:09 unknown user.warn kernel:   Normal          0 ->    16384
Jan  1 01:00:09 unknown user.warn kernel:   HighMem     16384 ->    16384
Jan  1 01:00:09 unknown user.info kernel: early_node_map[1] active PFN ranges
Jan  1 01:00:09 unknown user.warn kernel:     0:        0 ->    16384
Jan  1 01:00:09 unknown user.debug kernel: On node 0 totalpages: 16384
Jan  1 01:00:09 unknown user.debug kernel:   Normal zone: 0 pages used for memmap
Jan  1 01:00:09 unknown user.debug kernel:   Normal zone: 0 pages reserved
Jan  1 01:00:09 unknown user.debug kernel:   Normal zone: 16384 pages, LIFO batch:3
Jan  1 01:00:09 unknown user.debug kernel:   HighMem zone: 0 pages used for memmap
Jan  1 01:00:09 unknown user.warn kernel: Built 1 zonelists.  Total pages: 16384
Jan  1 01:00:09 unknown user.notice kernel: Kernel command line: root=/dev/mtdblock2 noinitrd console=ttyS0,115200
Jan  1 01:00:09 unknown user.warn kernel: Primary instruction cache 32kB, physically tagged, 4-way, linesize 32 bytes.
Jan  1 01:00:09 unknown user.warn kernel: Primary data cache 32kB, 4-way, linesize 32 bytes.
Jan  1 01:00:09 unknown user.info kernel: Synthesized TLB refill handler (20 instructions).
Jan  1 01:00:09 unknown user.info kernel: Synthesized TLB load handler fastpath (32 instructions).
Jan  1 01:00:09 unknown user.info kernel: Synthesized TLB store handler fastpath (32 instructions).
Jan  1 01:00:09 unknown user.info kernel: Synthesized TLB modify handler fastpath (31 instructions).
Jan  1 01:00:09 unknown user.info kernel: Setting up vectored interrupts
Jan  1 01:00:09 unknown user.warn kernel: PID hash table entries: 512 (order: 9, 2048 bytes)
Jan  1 01:00:09 unknown user.warn kernel: CPU: BCM5357 rev 2 pkg 10 at 500 MHz
Jan  1 01:00:09 unknown user.warn kernel: Using 250.000 MHz high precision timer.
Jan  1 01:00:09 unknown user.info kernel: console [ttyS0] enabled
Jan  1 01:00:09 unknown user.warn kernel: Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Jan  1 01:00:09 unknown user.warn kernel: Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Jan  1 01:00:09 unknown user.info kernel: Memory: 61372k/65536k available (32k kernel code, 4100k reserved, 2610k data, 120k init, 0k highmem)
Jan  1 01:00:09 unknown user.debug kernel: Calibrating delay loop... 249.85 BogoMIPS (lpj=1249280)
Jan  1 01:00:09 unknown user.warn kernel: Mount-cache hash table entries: 512
Jan  1 01:00:09 unknown user.info kernel: NET: Registered protocol family 16
Jan  1 01:00:09 unknown user.warn kernel: PCI: no core
Jan  1 01:00:09 unknown user.warn kernel: PCI: no core
Jan  1 01:00:09 unknown user.warn kernel: PCI: Fixing up bus 0
Jan  1 01:00:09 unknown user.info kernel: NET: Registered protocol family 2
Jan  1 01:00:09 unknown user.info kernel: Time: MIPS clocksource has been installed.
Jan  1 01:00:10 unknown user.warn kernel: IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
Jan  1 01:00:10 unknown user.warn kernel: TCP established hash table entries: 2048 (order: 2, 16384 bytes)
Jan  1 01:00:10 unknown user.warn kernel: TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
Jan  1 01:00:10 unknown user.info kernel: TCP: Hash tables configured (established 2048 bind 2048)
Jan  1 01:00:10 unknown user.info kernel: TCP reno registered
Jan  1 01:00:10 unknown user.info kernel: squashfs: version 3.0 (2006/03/15) Phillip Lougher
Jan  1 01:00:10 unknown user.info kernel: io scheduler noop registered (default)
Jan  1 01:00:10 unknown user.info kernel: HDLC line discipline: version $Revision: 4.8 $, maxframe=4096
Jan  1 01:00:10 unknown user.info kernel: N_HDLC line discipline registered.
Jan  1 01:00:10 unknown user.info kernel: Serial: 8250/16550 driver $Revision: 1.90 $ 2 ports, IRQ sharing disabled
Jan  1 01:00:10 unknown user.info kernel: serial8250: ttyS0 at MMIO 0xb8000300 (irq = 8) is a 16550A
Jan  1 01:00:10 unknown user.info kernel: PPP generic driver version 2.4.2
Jan  1 01:00:10 unknown user.info kernel: MPPE/MPPC encryption/compression module registered
Jan  1 01:00:10 unknown user.info kernel: NET: Registered protocol family 24
Jan  1 01:00:10 unknown user.info kernel: PPPoL2TP kernel driver, V0.18.3
Jan  1 01:00:10 unknown user.info kernel: PPTP driver version 0.8.5
Jan  1 01:00:10 unknown user.err kernel: pflash: found no supported devices
Jan  1 01:00:10 unknown user.warn kernel: Found an ST compatible serial flash with 256 64KB blocks; total size 16MB
Jan  1 01:00:10 unknown user.notice kernel: Creating 5 MTD partitions on "sflash":
Jan  1 01:00:10 unknown user.notice kernel: 0x00000000-0x00040000 : "pmon"
[... cut ...]
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:21 unknown user.info kernel: device eth2 left promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:21 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:21 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering forwarding state

pedro0311, 2025-03-04:
My mistake, I didn't explain exactly how to properly perform this update procedure and get the syslog.
I prepared another image: https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-E3200-K26MIPSR2_RTN-USB-NVRAM60K-2025.1.004-beta-test_upgrade-VPN.bin
Just upgrade it without wifi, then with wifi - and upload the syslog here.
The syslog:
- make sure your syslog is saved to a USB stick,
- make upgrade,
- after reboot, look at syslog for "STARTING UPGRADE" (the last one - you can also have it from previous upgrades),
- copy everything after that, up to the "/.../unknown syslog.info syslogd started: BusyBox v1.37.0" or something like that (I mean to the start of the new one after upgrade).

pedro0311, 2025-03-04:
And of course, please let me know if the upgrade with wifi turned on was successful.

txnative777, 2025-03-04:
I am not a mind reader. I kinda questioned, how is it going to be saved properly without a usb drive? No worries I can get back in no time.

Regards

txnative777, 2025-03-04:
Test build did not initialize or complete, the timer counter counted up to 227 seconds then stopped.

Jan  1 01:00:20 unknown user.info init[1]: Linksys E3200 v1.0: FreshTomato 2025.1.004 K26MIPSR2_RTN beta test_upgrade USB VPN
Jan  1 01:00:20 unknown user.notice kernel: Ebtables v2.0 registered
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth1 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth2 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:21 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering forwarding state
Jan  1 01:05:07 unknown daemon.debug httpd[804]: *** wi_upgrade: END: stop_syslog(), #remove_storage_main(), #stop_usb()
Jan  1 01:05:08 unknown syslog.info syslogd exiting
Jan  1 01:00:18 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:18 unknown user.info kernel: br0: port 3(eth2) entering forwarding state
Jan  1 01:00:18 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:18 unknown user.info kernel: br0: port 1(vlan1) entering forwarding state
Jan  1 01:00:18 unknown user.info init[1]: telnetd is started
Jan  1 01:00:18 unknown authpriv.info dropbear[468]: Running in background
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: started, version 2.91rc5 cachesize 4096
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth no-DNSSEC no-ID loop-detect no-inotify no-dumpfile
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:18 unknown daemon.info dnsmasq-dhcp[472]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:18 unknown daemon.warn dnsmasq[472]: failed to access /etc/resolv.dnsmasq: No such file or directory
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: read /etc/hosts - 3 names
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:19 unknown user.info init[1]: httpd is started
Jan  1 01:00:19 unknown user.info init[1]: starting rstats
Jan  1 01:00:19 unknown user.info init[1]: wireless roaming assistant is stopped
Jan  1 01:00:19 unknown user.info init[1]: MultiWAN: MWAN is 1 (max 4)
Jan  1 01:00:19 unknown user.warn kernel: vlan2: Setting MAC address to  20 aa 4b 00 28 47.
Jan  1 01:00:19 unknown user.debug kernel: vlan2: add 33:33:00:00:00:01 mcast address to master interface
Jan  1 01:00:19 unknown user.debug kernel: vlan2: add 01:00:5e:00:00:01 mcast address to master interface
Jan  1 01:00:20 unknown user.info kernel: IMQ starting with 2 devices...
Jan  1 01:00:20 unknown user.info kernel: IMQ driver loaded successfully.
Jan  1 01:00:20 unknown user.info kernel:       Hooking IMQ after NAT on PREROUTING.
Jan  1 01:00:20 unknown user.info kernel:       Hooking IMQ before NAT on POSTROUTING.
Jan  1 01:00:20 unknown user.info init[1]: Linksys E3200 v1.0: FreshTomato 2025.1.004 K26MIPSR2_RTN beta test_upgrade USB VPN
Jan  1 01:00:20 unknown user.notice kernel: Ebtables v2.0 registered
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth1 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth2 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:21 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:21 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering forwarding state
Jan  1 01:01:08 unknown daemon.debug httpd[570]: *** wi_upgrade: END: stop_syslog(), #remove_storage_main(), #stop_usb()
Jan  1 01:01:09 unknown user.notice kernel: klogd: exiting
Jan  1 01:01:09 unknown syslog.info syslogd exiting
Jan  1 01:00:17 unknown user.info hotplug[408]: USB vfat fs at /dev/sda1 mounted on /tmp/mnt/sda1
Jan  1 01:00:17 unknown user.debug kernel: vlan1: add 33:33:00:00:00:01 mcast address to master interface
Jan  1 01:00:17 unknown user.debug kernel: vlan1: add 01:00:5e:00:00:01 mcast address to master interface
Jan  1 01:00:17 unknown user.info kernel: vlan1: dev_set_allmulti(master, 1)
Jan  1 01:00:17 unknown user.info kernel: vlan1: dev_set_promiscuity(master, 1)
Jan  1 01:00:17 unknown user.info kernel: device eth0 entered promiscuous mode
Jan  1 01:00:17 unknown user.info kernel: device vlan1 entered promiscuous mode
Jan  1 01:00:17 unknown user.info kernel: device eth1 entered promiscuous mode
[... cut ...]
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth2 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:21 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering forwarding state

txnative777, 2025-03-04:
Is this a problem with 5GHz/wl not shutting down to begin the upgrade up until the end of the upgrading procedure?

I only mentioned this due to disabling wifi allows upgrading to follow through from beginning to end and that these are the only models to have these issues as well as the 40MHz side band issue with wireless mode rebooting in wireless client mode, wireless ethernet bridge mode using 5GHz IMO.

txnative777, 2025-03-04:
I performed a quick test by using the 2025.1 build from the freshtomato.org downloads and from the recent 004-beta-test build I used the wps_button to flash the 2025.1 build.

Once the E3200 was up and running, instead of using the wps_button I disabled only the 5GHz from status-overview allowing the upgrade back to the 004-beta-test build successfully using clear nvram during upgrade, this test was to isolate this to the bcm43236 5GHz only instead of both radios.

I assume somewhere if possible to add a down command on these partiticular models 5GHz, once the upgrading is started to finish both wifi interfaces should be down and become active once the boot routine begins for all interfaces to load as normal startup routne.

txnative777, 2025-03-04:
I feel that it was most likely misdiagnose or no one took that time to thoroughly troubleshoot it and only assumed in was nvram or just didn't bother with it, but that link from the forum I posted was about basically the issue now and that was back in 2021.

**On a separate note:
I built Mega-VPN and disabled 5GHz but I am in wireless ethernet bridge mode and wanted to test a dirty upgrade and as it has been working without 5GHz it took at least 180 seconds to flash and to come back online as designed**.

pedro0311, 2025-03-04:
Please test 005: https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-E3200-K26MIPSR2_RTN-USB-NVRAM60K-2025.1.005-beta-test_upgrade-VPN.bin
(also syslog).
It disables WL entirely during upgrade.

txnative777, 2025-03-04:
> Please test 005: https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-E3200-K26MIPSR2_RTN-USB-NVRAM60K-2025.1.005-beta-test_upgrade-VPN.bin (also syslog). It disables WL entirely during upgrade.

okay sounds like what needs to be done, be back in a few with results.

txnative777, 2025-03-04:
> > Please test 005: https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-E3200-K26MIPSR2_RTN-USB-NVRAM60K-2025.1.005-beta-test_upgrade-VPN.bin (also syslog). It disables WL entirely during upgrade.
> 
> okay sounds like what needs to be done, be back in a few with results.

Interesting that the upgrade didn't continue and was left with "Error reading file" but it did clear nvram and I had to manually  
type the ip addr to look over status-overview.

The gui counted up to 75 seconds then the page went to a generic error, I started to ping my ip addr that I was using for wireless ethernet bridge until it stopped around 110 seconds.

I then cleared the ip from debian network manager, then noticed FreshTomato ssids were available then debian received the the ip from dhcp and that was it for the test.

txnative777, 2025-03-04:
I have also the WNDR3400v2 and Belkin N600 DB Wireless N+ but the results may end up the same, but if you like to build one for both is up to you?

Regards

pedro0311, 2025-03-05:
May I see a syslog from upgrade 005 -> 005 with WL enabled (in this one you should find "STARTING UPGRADE")?

txnative777, 2025-03-05:
> May I see a syslog from upgrade 005 -> 005 with WL enabled (in this one you should find "STARTING UPGRADE")?

**Of course but just to also mention this time I re ran the upgrade as a dirty upgrade with the usb drive attached, the upgrade went without any problem which is great. 

Could you build a wndr3400v2-vpn build just to check if the E3200 0005-beta wasn't a fluke, meaning it took two upgrade flash to see proper results as it should had worked the first time IMHO.**

Jan  1 01:00:20 unknown user.info init[1]: Linksys E3200 v1.0: FreshTomato 2025.1.005 K26MIPSR2_RTN beta test_upgrade USB VPN
Jan  1 01:00:20 unknown user.notice kernel: Ebtables v2.0 registered
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth1 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth2 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:21 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:21 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering forwarding state
Jan  1 01:02:06 unknown user.info init[1]: *** exec_service: STARTING UPGRADE
Jan  1 01:02:07 unknown user.info init[1]: *** exec_service: AFTER stop_sched()
Jan  1 01:02:09 unknown user.info init[1]: *** exec_service: AFTER stop_cron()
Jan  1 01:02:10 unknown user.info init[1]: *** exec_service: AFTER restart_nas_services()
Jan  1 01:02:11 unknown user.info init[1]: *** exec_service: AFTER rstats()
Jan  1 01:02:11 unknown user.info init[1]: *** exec_service: AFTER cstats()
Jan  1 01:02:12 unknown user.info init[1]: *** exec_service: AFTER buttons()
Jan  1 01:02:13 unknown user.info init[1]: *** exec_service: AFTER stop_upnp()
Jan  1 01:02:14 unknown user.info init[1]: *** exec_service: AFTER stop_wan(1)
Jan  1 01:02:14 unknown user.notice adblock[835]: Stopping Adblock/DNS-filtering...
Jan  1 01:02:15 unknown user.info init[1]: *** exec_service: AFTER stop_adblock()
Jan  1 01:02:15 unknown user.notice adblock[885]: Stopping Adblock/DNS-filtering...
Jan  1 01:02:16 unknown user.info init[1]: *** exec_service: AFTER remove_conntrack()
Jan  1 01:02:17 unknown user.info init[1]: *** exec_service: AFTER stop_zebra()
Jan  1 01:02:18 unknown user.info init[1]: *** exec_service: AFTER stop_jffs2()
Jan  1 01:02:19 unknown user.notice kernel: klogd: exiting
Jan  1 01:02:19 unknown syslog.info syslogd exiting
Jan  1 01:00:17 unknown user.info hotplug[408]: USB vfat fs at /dev/sda1 mounted on /tmp/mnt/sda1
Jan  1 01:00:17 unknown user.debug kernel: vlan1: add 33:33:00:00:00:01 mcast address to master interface
Jan  1 01:00:17 unknown user.debug kernel: vlan1: add 01:00:5e:00:00:01 mcast address to master interface
Jan  1 01:00:17 unknown user.info kernel: vlan1: dev_set_allmulti(master, 1)
Jan  1 01:00:17 unknown user.info kernel: vlan1: dev_set_promiscuity(master, 1)
Jan  1 01:00:17 unknown user.info kernel: device eth0 entered promiscuous mode
Jan  1 01:00:17 unknown user.info kernel: device vlan1 entered promiscuous mode
Jan  1 01:00:17 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:18 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:18 unknown user.info kernel: br0: port 3(eth2) entering forwarding state
Jan  1 01:00:18 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:18 unknown user.info kernel: br0: port 1(vlan1) entering forwarding state
Jan  1 01:00:18 unknown user.info init[1]: telnetd is started
Jan  1 01:00:18 unknown authpriv.info dropbear[468]: Running in background
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: started, version 2.91rc5 cachesize 4096
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth no-DNSSEC no-ID loop-detect no-inotify no-dumpfile
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:18 unknown daemon.info dnsmasq-dhcp[472]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:18 unknown daemon.warn dnsmasq[472]: failed to access /etc/resolv.dnsmasq: No such file or directory
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: read /etc/hosts - 3 names
Jan  1 01:00:18 unknown daemon.info dnsmasq[472]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:19 unknown user.info init[1]: httpd is started
Jan  1 01:00:19 unknown user.info init[1]: starting rstats
Jan  1 01:00:19 unknown user.info init[1]: wireless roaming assistant is stopped
Jan  1 01:00:19 unknown user.info init[1]: MultiWAN: MWAN is 1 (max 4)
Jan  1 01:00:19 unknown user.warn kernel: vlan2: Setting MAC address to  20 aa 4b 00 28 47.
Jan  1 01:00:19 unknown user.debug kernel: vlan2: add 33:33:00:00:00:01 mcast address to master interface
Jan  1 01:00:19 unknown user.debug kernel: vlan2: add 01:00:5e:00:00:01 mcast address to master interface
Jan  1 01:00:19 unknown user.info kernel: IMQ starting with 2 devices...
Jan  1 01:00:19 unknown user.info kernel: IMQ driver loaded successfully.
Jan  1 01:00:19 unknown user.info kernel:       Hooking IMQ after NAT on PREROUTING.
Jan  1 01:00:19 unknown user.info kernel:       Hooking IMQ before NAT on POSTROUTING.
Jan  1 01:00:20 unknown user.info init[1]: Linksys E3200 v1.0: FreshTomato 2025.1.005 K26MIPSR2_RTN beta test_upgrade USB VPN

pedro0311, 2025-03-05:
Let's make one more test with E3200: https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-E3200-K26MIPSR2_RTN-USB-NVRAM60K-2025.1.006-beta-test_upgrade-VPN.bin
This one with no logs, but (hopefully) should works with WL enabled.

txnative777, 2025-03-05:
> Let's make one more test with E3200: https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-E3200-K26MIPSR2_RTN-USB-NVRAM60K-2025.1.006-beta-test_upgrade-VPN.bin This one with no logs, but (hopefully) should works with WL enabled.

Okay, it shouldn't long here.

txnative777, 2025-03-05:
I set to clear nvram this time with the upgrade, the result is that it went accordingly as design. At least 60 seconds to initialize then 300 seconds to finish with counter.

Great!

pedro0311, 2025-03-05:
Of course I can build another image, just let me know which one.

BTW, I see a problem here, because we need to add some warning or something like that when a user wants to update such a router (USBAP) via Wi-Fi.

txnative777, 2025-03-05:
> Of course I can build another image, just let me know which one.
> 
> BTW, I see a problem here, because we need to add some warning or something like that when a user wants to update such a router (USBAP) via Wi-Fi.

The **Netgear WNDR3400v2**.

Yes I had the same thought as well on this upgrade issue, to at least have some small documentation for these models.

The question is where? Perhaps the wiki pointing to a link in forum or just a small mention in the Wiki for these USB models?

pedro0311, 2025-03-05:
https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-WNDR3400v2-K26MIPSR2_RTN-USB-NVRAM64K-2025.1.006-beta-test_upgrade-VPN.chk

I think that such warning should be added to upgrade page for all USBAP routers.

txnative777, 2025-03-05:
> https://bitbucket.org/pedro311/freshtomato-mips/downloads/freshtomato-WNDR3400v2-K26MIPSR2_RTN-USB-NVRAM64K-2025.1.006-beta-test_upgrade-VPN.chk
> 
> I think that such warning should be added to upgrade page for all USBAP routers.

indeed, should we added to the wiki links to specific threads? I'll test the this netgear image and be back with results.

txnative777, 2025-03-05:
> indeed, should we added to the wiki links to specific threads? I'll test the this netgear image and be back with results.

within the admin-upgrade?

The upgrade didn't work the first time same result as the E3200 but after I re ran the upgrade after a few moments and did the upgrade and worked as it did for E3200.

Average time for WNDR3400v2 at least this model takes at least 4 mins and 20 seconds a bit shorter time than the E3200 and the Belkin usually about the same time as E3200.

Does this mean to use wps to disable wifi so the upgrade from previous builds to fixed builds will be then loaded, then after upgrades should go as planned?

pedro0311, 2025-03-05:
> within the admin-upgrade?

I think so.

> Does this mean to use wps to disable wifi so the upgrade from previous builds to fixed builds will be then loaded, then after upgrades should go as planned?

Exactly.

txnative777, 2025-03-05:
> > within the admin-upgrade?
> 
> I think so.
> 
> > Does this mean to use wps to disable wifi so the upgrade from previous builds to fixed builds will be then loaded, then after upgrades should go as planned?
> 
> Exactly.

Agreed, once it is flashed this upgrade issue for these models should help those users transition from past to present to work as normal once this fix is done.

txnative777, 2025-03-05:
I did a dirty flash from 2024.5 to your fix image.
The upgrade took about 5 mins 20 seconds to complete.
upgrade counter continued all the way to 3:30 seconds, then after transitioned to dark screen until 95 second count down to complete the dirty upgrade.

I think I should've gone back further like a 2023.1 and do a 30-30-30 then repeat the steps to ensure what we are seeing is accurate enough or are we satisfied with current results?

Jan  1 01:00:20 unknown user.info init[1]: Netgear WNDR3400v2: FreshTomato 2025.1.006 K26MIPSR2_RTN beta test_upgrade USB VPN
Jan  1 01:00:20 unknown user.notice kernel: Ebtables v2.0 registered
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth1 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth2 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:21 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:21 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering forwarding state
Jan  1 01:00:27 unknown user.notice adblock[621]: Stopping Adblock/DNS-filtering...
Jan  1 01:00:27 unknown user.notice kernel: klogd: exiting
Jan  1 01:00:27 unknown syslog.info syslogd exiting
Jan  1 01:00:16 unknown user.info hotplug[432]: USB vfat fs at /dev/sda1 mounted on /tmp/mnt/sda1
Jan  1 01:00:16 unknown user.debug kernel: vlan1: add 33:33:00:00:00:01 mcast address to master interface
Jan  1 01:00:16 unknown user.debug kernel: vlan1: add 01:00:5e:00:00:01 mcast address to master interface
Jan  1 01:00:16 unknown user.info kernel: vlan1: dev_set_allmulti(master, 1)
Jan  1 01:00:16 unknown user.info kernel: vlan1: dev_set_promiscuity(master, 1)
Jan  1 01:00:16 unknown user.info kernel: device eth0 entered promiscuous mode
Jan  1 01:00:16 unknown user.info kernel: device vlan1 entered promiscuous mode
Jan  1 01:00:17 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:17 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:17 unknown user.info kernel: br0: port 3(eth2) entering forwarding state
Jan  1 01:00:17 unknown user.info init[1]: telnetd is started
Jan  1 01:00:17 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:17 unknown user.info kernel: br0: port 1(vlan1) entering forwarding state
Jan  1 01:00:17 unknown authpriv.info dropbear[492]: Running in background
Jan  1 01:00:18 unknown daemon.info dnsmasq[496]: started, version 2.91rc5 cachesize 4096
Jan  1 01:00:18 unknown daemon.info dnsmasq[496]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth no-DNSSEC no-ID loop-detect no-inotify no-dumpfile
Jan  1 01:00:18 unknown daemon.info dnsmasq[496]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:18 unknown daemon.info dnsmasq-dhcp[496]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:18 unknown daemon.warn dnsmasq[496]: failed to access /etc/resolv.dnsmasq: No such file or directory
Jan  1 01:00:18 unknown daemon.info dnsmasq[496]: read /etc/hosts - 3 names
Jan  1 01:00:18 unknown daemon.info dnsmasq[496]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:19 unknown user.info init[1]: httpd is started
Jan  1 01:00:19 unknown user.info init[1]: starting rstats
Jan  1 01:00:19 unknown user.info init[1]: wireless roaming assistant is stopped
Jan  1 01:00:19 unknown user.info init[1]: MultiWAN: MWAN is 1 (max 4)
Jan  1 01:00:19 unknown user.warn kernel: vlan2: Setting MAC address to  84 1b 5e 43 46 e3.
Jan  1 01:00:19 unknown user.debug kernel: vlan2: add 33:33:00:00:00:01 mcast address to master interface
Jan  1 01:00:19 unknown user.debug kernel: vlan2: add 01:00:5e:00:00:01 mcast address to master interface
Jan  1 01:00:19 unknown user.info kernel: IMQ starting with 2 devices...
Jan  1 01:00:19 unknown user.info kernel: IMQ driver loaded successfully.
Jan  1 01:00:19 unknown user.info kernel:       Hooking IMQ after NAT on PREROUTING.
Jan  1 01:00:19 unknown user.info kernel:       Hooking IMQ before NAT on POSTROUTING.
Jan  1 01:00:20 unknown user.info init[1]: Netgear WNDR3400v2: FreshTomato 2025.1.006 K26MIPSR2_RTN beta test_upgrade USB VPN
Jan  1 01:00:20 unknown user.notice kernel: Ebtables v2.0 registered
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth1 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 2(eth1) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:20 unknown user.info kernel: device eth2 left promiscuous mode
Jan  1 01:00:20 unknown user.info kernel: br0: port 3(eth2) entering disabled state
Jan  1 01:00:21 unknown user.info kernel: device eth1 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 2(eth1) entering forwarding state
Jan  1 01:00:21 unknown user.info kernel: device eth2 entered promiscuous mode
Jan  1 01:00:21 unknown user.info kernel: br0: port 3(eth2) entering forwarding state
Jan  1 01:03:25 unknown auth.info login[886]: root login on 'pts/0'

pedro0311, 2025-03-06:
I will close it with commit.

pedro0311, 2025-03-06:
@txnative777 : before we go any further. 

You mentioned on LI that this problem didn't occur before (apart from the described case with version 2021.3, later fixed). Could you downgrade to the oldest version available online, i.e. 2022.1, and then try to upgrade to 2022.2 without turning off wi-fi?
Does this version (2022.1) work?

txnative777, 2025-03-06:
> [@txnative777](https://github.com/txnative777) : before we go any further.
>
> You mentioned on LI that this problem didn't occur before (apart from the described case with version 2021.3, later [...]

I can check and depsite being on different time zones I will get the information shortly, currently right now I did flash from 2025.1 to 2022.1, cleared nvram before flash from 2022.1 to 2022.2 from wireless ethernet bridge config by performing a dirty flash and it seem to work but it was unusually longer like 5 mins 23 seconds.

I used my **Belkin N600 DB Wireless N+** for this test.

I downloaded the 2025.1 and dirty flashed from 2022.2 to 2025.1 except this time I wanted to clear nvram and again it took a little over 5 mins then google timed out, refreshing the screen brought up the google generic secure site screen and as I clicked it went to the Error reading file.

Clicking pass it brings you back to admin-upgrade, already ready witht the check mark to clear nvram with the 2025.1 still ready.

pedro0311, 2025-03-10:
Let's stick to just one router model because otherwise we'll never know what's going on and which FW actually allows the upgrade and which doesn't.
BTW, this longer upgrade time may be due to hardware differences between the E3200 and N600.
I see it like this: let's try an upgrade (E3200 for consistency) 2022.1 -> 2022.2: what's the result?
Then e.g. 2022.5 -> 2022.6: was the upgrade successful?
Etc...

Since the older FW allows for an upgrade with WL enabled, you should find the last working FW and then look for the changes that cause these problems. It will take some time, but I don't see any other way.

txnative777, 2025-03-10:
> Let's stick to just one router model because otherwise we'll never know what's going on and which FW actually allows the upgrade and which doesn't. BTW, this longer upgrade time may be due to [...]

Using the E3200, I did go back to 2022.1 from the beta image that is working, cleared nvram even 30-30-30 except this time dirty upgrading from wireless ethernet bridge config to 2022.2 this test was to dirty upgrade to 2022.2 but this this time after a full 5 mins the uprade screen stopped while using google I had refreshed the page this brought it back to the "Error reading file".

This also occurs trying to upgrade between all builds from 2022.1 leading up to 2022.6 which turns to VPN image, in the previous builds it was 32K-Mega-VPN builds, no matter selecting clear nvram or allowing a dirty upgrade the result was the same.

Timing out, Error reading file, returning to the admin-upgrade with the same settings I had setup for the upgrade. The tests were performed using wireless ethernet bridge and no config settings, I did also clear nvram before performng each tests.

txnative777, 2025-03-11:
> > OK, so now we know that even 2022.1 has this issue. Hmmm....
> > I don't know if it makes sense to go back even further in versions and look for a working one - in my opinion it's a waste of time. It's better to do our workaround with disabling WL completely during the upgrade + adding a warning for routers with wl_high on admin-upgrade page that you should only upgrade via cable. What do you think about it?
> 
> I agree with your reply, at least only for this upcoming release a **Warning can be set for these models**. then after upgrading should go as designed.
> 
> I'll upgrade to the build that was successfull for the E3200 to test and check if I can upgrade wirelessly and report back.

**Silly me of course wifi gets disabled** but on the flip side it works only through lan connected unless somehow the image gets passed through the upgrade quickly through wifi then once the upgrade initializes wifi becomes disabled, but what you have done is good enough for these 5GHz routers @pedro0311. The suggested warning should good enough then after it shouldn't be necessary in future builds.

If you have other builds to try let me know but this **Open ticket** can be closed.

Regards

pedro0311, 2025-03-12:
> The suggested warning should good enough then after it shouldn't be necessary in future builds.

Nope. If we do it this way, it should always be there, even in feature releases, because we need WL be disabled during upgrade every time now.
I'll look into this later, but now I have a more important issue, which affects all routers (ARM + MIPS), because of dnsmasq crash: https://github.com/FreshTomato-Project/freshtomato-mips/issues/6#issuecomment-2717568669

BR

HommeOursPorc, 2025-03-12:
> Trying again, the timer will count down to 2:24 or so seconds then the page goes blank. During both tries the firmware did flash but failed to goto the next screen of counting down page of successful flash.

FWIW, I'm experiencing this exact problem when upgrading my RT-N12 D1. Same thing, I see the upgrade chronometer/incremental timer, then it goes blank. No message about a successful flash, no reboot countdown, nothing. My browser just won't load the page. The worst is that the router hang. 

It just happened right now. The erase nvram checkbox was selected. Upgrading from `freshtomato-K26MIPSR2_RTN-NVRAM32K-2024.4-Max.trx` to `freshtomato-K26MIPSR2_RTN-NVRAM32K-2025.1-Max.trx`.

The upgrade was successful, but the router hanged after the upgrade and required a manual power cycle to get it back. When it hangs, it still broadcast the wifi, but I can't connect to it at all. I didn't try with a Ethernet cable though.

pedro0311, 2025-03-14:
@txnative777 : I have a challenge for you. Could you flash an E3200 with an old version of tomato v140 (of course required nvram clear and Ctrl+F5 on upgrade page) (https://tomato.groov.pl/download/K26RT-N/build5x-140-MultiWAN/Linksys%20E-series/)?
Then try with WL enabled, flash again to v140 and see if it completes correctly?
Thanks!

pedro0311, 2025-03-14:
@txnative777 @HommeOursPorc : 
One question, because I'm not entirely sure: am I to understand that in the case of routers with wl_high the upgrade process itself is correct (FW is updated) and there is only a problem with displaying the page with the countdown, or is there something else? And if so, what is it?

HommeOursPorc, 2025-03-14:
> [@txnative777](https://github.com/txnative777) [@HommeOursPorc](https://github.com/HommeOursPorc) : One question, because I'm not entirely sure: am I to understand that in the case of routers with [...]

I'm not sure what is wl_high, does my RT-N12 have this? 

I experienced the problem many times, when it happens:
  - router is flashed successfully
  - no successful message, reboot message or countdown is displayed after the upgrade chronometer. The "last" page before the device reboot is not displayed. Browser say the page does not exist or something similar.
  - in my case, the router does NOT reboot, the router hands and need a manual power cycle. 
  - once powered back ON, the router works correctly and seems to be correctly upgraded. 

I hope the helps 🙂

HommeOursPorc, 2025-03-14:
Could the problem be somewhere over there: 

https://github.com/FreshTomato-Project/freshtomato-mips/blob/0fc276d927fc6e8a2ab119b5b3f71b2e57963820/release/src/router/httpd/upgrade.c

```
void wo_flash(char *url)
{
	if (rboot) {
		sleep(1);
		parse_asp("/tmp/reboot.asp");
		web_close();

		if (nvram_get_int("remote_upgrade")) {
			killall("xl2tpd", SIGTERM);
			killall("pppd", SIGTERM);
		}

		sleep(2);

		sync();
		//kill(1, SIGTERM);
		reboot(RB_AUTOBOOT);

		exit(0);
	}

	parse_asp("error.asp");
}
```

parse_asp("/tmp/reboot.asp");

Or

web_close();

Look like the reboot asp page is not loaded properly and then it hangs and never proceed to the reboot. 

Maybe?

pedro0311, 2025-03-14:
I'm not interested in other routers than these ones with wl_high (USBAP): https://github.com/FreshTomato-Project/freshtomato-mips/blob/mips-RT-AC/release/src-rt/targets.mak#L176
I know about this problem, but even @shibby didn't know how to fix it.
So, my question is only about wl_high routers.

txnative777, 2025-03-14:
> [@txnative777](https://github.com/txnative777) : I have a challenge for you. Could you flash an E3200 with an old version of tomato v140 (of course required nvram clear and Ctrl+F5 on upgrade page) [...]

The 140-Mega-VPN did not complete the task of upgrading after 1 min 24 seconds the page was dropped and was unable to respond even with refreshing the page. I resorted to editing clearing the address up from the end of the uprade cgi stuff and then typing /admin-upgrade.asp to reach the page with no sign of the image the page as cleared.

pedro0311, 2025-03-17:
OK, so the problem has been there from the very beginning.

txnative777, 2025-03-17:
> OK, so the problem has been there from the very beginning.

It appears so that it most likely was not reported or just overlooked.

pedro0311, 2025-03-17:
At least now I'm sure that the only possible solution (or rather workaround) is to disable WL during the update on these routers and add a warning about this fact..

commit 454ef658 (2024-12-30): UPnP IGD & PCP: Add `Expires` to port forward listing

Close #5
commit d1a324dc (2025-03-17): build: disable Wi-Fi during upgrade for routers with wl_high (USBAP) module to make it work correctly, and add a warning about using only cable connection during this process (close #5).
closed 2025-03-17 by pedro0311