FreshTomato-Project/freshtomato-mips #6: dnsmasq DHCP server crash or close itself after assigning one IP address

HommeOursPorc, 2025-03-08:
Router: RT-N12 D1
Firmware: freshtomato-K26MIPSR2_RTN-NVRAM32K-2025.1-Max.trx

Fresh upgrade, nvram have been cleared before and after and many times after.

Problem: Client device connecting wirelessly to freshly booted router correctly receive IP address. SSH into the router and dnsmasq is NOT running. 

Disconnecting the device, then trying to reconnect. It is impossible to reconnect, DHCP server is NOT running and the client device won't received an IP. 

So dnsmasq assign one IP address, then crash or close itself.

Using a static IP in client device settings bypass the problem and let me connect to the router wirelessly. 

I don't see anything in the logs.

```
root@unknown:/tmp/home/root# ps | grep dnsmasq
  805 root      1368 R    grep dnsmasq
root@unknown:/tmp/home/root# ps
  PID USER       VSZ STAT COMMAND
    1 root      1380 S    /sbin/init noinitrd
    2 root         0 SW<  [kthreadd]
    3 root         0 SW<  [ksoftirqd/0]
    4 root         0 SW<  [events/0]
    5 root         0 SW<  [khelper]
   18 root         0 SW<  [kblockd/0]
   36 root         0 SW   [pdflush]
   37 root         0 SW   [pdflush]
   38 root         0 SW<  [kswapd0]
   39 root         0 SW<  [aio/0]
   81 root         0 SW<  [mtdblockd]
  250 root       636 S    hotplug2 --persistent --no-c
  286 root      1368 S    buttons
  289 root      1336 S    console
  290 root      1380 S    /bin/sh
  297 root      1372 S    syslogd -L -s 50 -b 1 -l 8
  299 root      1372 S    klogd
  350 root      1376 R    telnetd -p 23
  353 root      1136 S    dropbear -p 22 -a
  361 root      3600 S    httpd
  363 root      1384 S    crond -l 9
  367 root      1236 S    rstats
  413 root      1380 S    udhcpc -i vlan1 -b -s /sbin/
  722 root      1384 S    -sh
  806 root      1376 R    ps
```

HommeOursPorc, 2025-03-08:
Forgot to mention. This is with the default settings with FreshTomato24 open wifi (no security).

HommeOursPorc, 2025-03-09:
Just tested on MIPS RT-N66U, same issue. Changed SSID and added security, issue persist. Both router are in my "lab" and doesn't use any WAN. Anyone else experiencing this?

txnative777, 2025-03-09:
> Router: RT-N12 D1 Firmware: freshtomato-K26MIPSR2_RTN-NVRAM32K-2025.1-Max.trx
>
> Fresh upgrade, nvram have been cleared before and after and many times after.
>
> Problem: Client device connecting [...]

I have been experience these issues since 2023 builds but it may go back further but to understand your open ticket better,  after flashing and clearing nvram does the client have an ip first, then after if you clear or reset the client's ip, your router is no longer sending dhcp to your clients wireless even if the client is connected through lan port?

ARM doesn't appear to have this effect and since I usually test on my other mips models I have been experience a dnsmasq dhcp issue and although I can use service to start dnmsasq using a static ip, repeating those steps of releasing the ip from a client the I am unable to establish an ip from the either my E3200, WNDR3400v2 or F9K1102v3.

**You could post your syslog be sure to set up in admin-debug -> Enable segfault logging and set Kernel printk log level to Debug,** on a fresh session then post it.

If pedro311 needs one from me I can post one as well.

HommeOursPorc, 2025-03-09:
> I have been experience these issues since 2023 builds but it may go back further but to understand your open ticket better, after flashing and clearing nvram does the client have an ip first, then [...]

Yes, but I didn't use any wired LAN device, wireless only. It is exactly as you say, erased nvram (thorough), then once the router is booted, the wireless device (client, my phone in my case) correctly get assigned an IP, then if I turn off my phone wifi and turn it back on and try to reconnect to the FreshTomato24 open wifi I'm not getting any IP. Using a static IP I can connect to the router. Once logged in the router, I check the running process with `ps` and there's no dnsmasq process running. 

So dnsmasq must be running and then crash after issuing one IP address.

> ARM doesn't appear to have this effect and since I usually test on my other mips models I have been experience a dnsmasq dhcp issue and although I can use service to start dnmsasq using a static ip, [...]

I already sent the logs to Pedro in a private message, but I didn't include any extra debugging, only the normal logs. I'll check this out 🙂

pedro0311, 2025-03-10:
Simon added 2 more commits:
https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=9df1bd0cc1580fe3d2dd2c0fd21050e528d178c5
https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=e427d4b0e6223675cae8b17af81fdcf9fb7fc580
so I can build images for you, just let me know which ones exactly.

HommeOursPorc, 2025-03-10:
I will try to get you the log soon 🙂

I'm pretty sure I had the issue in 2024.5 too.

I can test on RT-N12 and/or RT-N66U, so I need images: 
freshtomato-RT-N66U-K26MIPSR2_RTN-USB-2025.1-AIO-64K

And/or

freshtomato-K26MIPSR2_RTN-NVRAM32K-2025.1-Max

pedro0311, 2025-03-10:
OK, I'll prepare freshtomato-K26MIPSR2_RTN-NVRAM32K-2025.1-Max - it builds faster ;)

HommeOursPorc, 2025-03-10:
> **You could post your syslog be sure to set up in admin-debug -> Enable segfault logging and set Kernel printk log level to Debug,** on a fresh session then post it.

Where are those logs saved? With the normal log? Any special path/location?

pedro0311, 2025-03-10:
So let me get this straight: does this issue only occur in non-WAN mode, or always? With the default configuration with DHCP enabled, right?

pedro0311, 2025-03-10:
syslog ;)

pedro0311, 2025-03-10:
Did you enable: 

> You could post your syslog be sure to set up in admin-debug -> Enable segfault logging and set Kernel printk log level to Debug, on a fresh session then post it.

HommeOursPorc, 2025-03-10:
Yes it was already set. First thing I did after cleaning nvram. Sorry I didn't write it above.

[image]

Will reboot and post logs again. Let the router run, the log is currently filled with ntpd errors, exactly like the one I posted above.

HommeOursPorc, 2025-03-10:
Here are the logs. I would've though the options above would add more lines in the logs 🤔

[syslog1.txt](https://github.com/user-attachments/files/19168069/syslog1.txt)

txnative777, 2025-03-10:
Not to change the flow of information, I noticed that while testing E3200 2022.x images I also checked if this was occurring, after clearing nvram on my tests for the other issue on 5GHz usb routers the E3200 was able to assign the ip after release the ip from Debian 12 networkmanager and waited 10 seconds, the result was the router assigned the ip, now this doesn't work on newer builds AFAIK?

HommeOursPorc, 2025-03-10:
> Just helping out, I may have found that 204.5 is when this issue started

I can confirm. Just tested 2024.4 on RT-N12 and the DHCP server function normally. I can also see the process in the list 🙂

So the problem start in 2024.5 and up.

[image]

>  "**service dnsmasq start**" it will assign the ip.

Doh! I never thought about trying the service 🤦
Good idea! Thanks!

HommeOursPorc, 2025-03-10:
@txnative777 Hello, can you check if connecting and enabling WAN correct the problem? 

I think the problem is with ntpd that keep on crashing...

Once I set a WAN connection, the problem stop. DHCP seems to work correctly now. 

Or maybe it is something else. Here what I did on my RT-N66U: 
1. Set WAN to static
2. Set WAN to Wireless Client and configured wl0
3. Set IP, gateway and DNS to point to my main router (192.168.1.1)
4. Changed LAN0 br0 to 192.168.66.1 and set DHCP ON for range 192.168.66.100-200
5. Set wl1 as Access Point with WPA2 PSK security
6. Save
7. Reboot

Now DHCP works correctly and the process is visible in the ps list. 

Router now have access to Internet and ntpd stopped crashing and router have proper time. Now DHCP work... Is ntpd the problem??

txnative777, 2025-03-11:
Were you originally trying to get your Asus connected for some config using wan then after is stated in your previous post?

In my situation, the router is not being setup initially and I noticed after releasing the ip on the client side then I am not able to receive an ip from the router not more than 15 seconds or so, but it is safe this started on build 2024.5.

pedro0311, 2025-03-11:
> Steps I took:
>
>   Cleared nvram (thorough)
>   Tried to connect to FreshTomato24 open wifi, no go. Can't connect. Phone complain it doesn't receive an IP address.

I have one RT-N12 (4MB) so I'll test it.
What was WAN config in your case?
I can set it as DHCP to upstream router by cable.

HommeOursPorc, 2025-03-11:
> > Steps I took:
> > Cleared nvram (thorough)
> > Tried to connect to FreshTomato24 open wifi, no go. Can't connect. Phone complain it doesn't receive an IP address.
> 
> I have one RT-N12 (4MB) so I'll test it. What was WAN config in your case? I can set it as DHCP to upstream router by cable.
> 
> //EDIT: should I set something more?

In my original case, no WAN and no wired LAN is connected and the router use the default settings, right after cleaning nvram.

I was only using wireless, no wire where involved 😋

No additional setting are needed.

HommeOursPorc, 2025-03-11:
> Were you originally trying to get your Asus connected for some config using wan then after is stated in your previous post?
>
> In my situation, the router is not being setup initially and I noticed [...]

See my last reply (answering Pedro), the problem occur with the default settings with no wire connected to the router.

pedro0311, 2025-03-11:
OK guys, it's not only MIPS related:
```
(WAN down)
Jan  1 01:00:15 unknown daemon.info dnsmasq[959]: started, version 2.91rc5-e427d4b cachesize 4096
Jan  1 01:00:15 unknown daemon.info dnsmasq[959]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:15 unknown daemon.info dnsmasq[959]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:15 unknown daemon.info dnsmasq-dhcp[959]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:15 unknown daemon.info dnsmasq[959]: read /etc/hosts - 3 names
Jan  1 01:00:15 unknown daemon.info dnsmasq[959]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:22 unknown daemon.info dnsmasq-dhcp[959]: DHCPDISCOVER(br0) 
Jan  1 01:00:22 unknown daemon.info dnsmasq-dhcp[959]: DHCPOFFER(br0)
Jan  1 01:00:22 unknown daemon.info dnsmasq-dhcp[959]: DHCPREQUEST(br0)
Jan  1 01:00:22 unknown daemon.info dnsmasq-dhcp[959]: DHCPACK(br0)		<-------- for my laptop
Jan  1 01:00:23 unknown kern.warn kernel: dnsmasq/959: potentially unexpected fatal signal 11.
Jan  1 01:00:23 unknown kern.warn kernel: Pid: 959, comm:              dnsmasq
Jan  1 01:00:23 unknown kern.warn kernel: CPU: 0    Tainted: P             (2.6.36.4brcmarm #2)
Jan  1 01:00:23 unknown kern.warn kernel: PC is at 0x2ce98
Jan  1 01:00:23 unknown kern.warn kernel: LR is at 0x2ce5c
Jan  1 01:00:23 unknown kern.warn kernel: pc : [<0002ce98>]    lr : [<0002ce5c>]    psr: 60000010
Jan  1 01:00:23 unknown kern.warn kernel: sp : be86ba60  ip : 0009384c  fp : 00000029
Jan  1 01:00:23 unknown kern.warn kernel: r10: 00000000  r9 : 00091e3c  r8 : 0000001a
Jan  1 01:00:23 unknown kern.warn kernel: r7 : 0000000e  r6 : 00093828  r5 : 00091e3c  r4 : 00000000
Jan  1 01:00:23 unknown kern.warn kernel: r3 : 00092c18  r2 : 00000000  r1 : 0004b754  r0 : 00000000
Jan  1 01:00:23 unknown kern.warn kernel: Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Jan  1 01:00:23 unknown kern.warn kernel: Control: 10c53c7d  Table: 9d774059  DAC: 00000015
(so dnsmasq is down)
/.../
Jan  1 01:00:39 unknown user.info preinit[1]: Asus RT-N18U: FreshTomato 2025.1 K26ARM USB AIO-64K-NOSMP
/.../
(WAN up)
Jan  1 01:00:55 unknown daemon.info dnsmasq[1480]: started, version 2.91rc5-e427d4b cachesize 4096
Jan  1 01:00:55 unknown daemon.info dnsmasq[1480]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:55 unknown daemon.info dnsmasq[1480]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:55 unknown daemon.info dnsmasq-dhcp[1480]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:55 unknown daemon.info dnsmasq[1480]: reading /etc/resolv.dnsmasq
Jan  1 01:00:55 unknown daemon.info dnsmasq[1480]: read /etc/hosts - 3 names
Jan  1 01:00:55 unknown daemon.info dnsmasq[1480]: read /etc/hosts.dnsmasq - 5 names
/.../
(so it's up, because of restart when WAN is up - if no WAN, no dnsmsaq...)
```

Working on it...

pedro0311, 2025-03-11:
Revert of @lancethepants commit didn't help...

pedro0311, 2025-03-11:
Ok, I have no time and desire to check what is wrong with latest dnsmasq, so @all ask @simon (https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=summary) what he broke, otherwise I'll revert to 6d730a2 or latest working for us.
Cheers.

HommeOursPorc, 2025-03-11:
I've registered for the mailing list but got no confirmation email yet. It's been an hour... Can someone tell me if there's an other way to file a bug report for dnsmasq? Should I email Simon directly?

Just to confirm, this seems to be the dnsmasq version that is used in 2024.4
```
Dnsmasq version 2.90-f006be7  Copyright (c) 2000-2024 Simon Kelley
Compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-scripts no-TFTP no-conntrack ipset no-nftset Tomato-helper no-auth no-cryptohash no-DNSSEC no-ID loop-detect no-inotify no-dumpfile
```
Is that correct?

Thanks

pedro0311, 2025-03-11:
Images with -
this one (2024-10-04) work: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=f006be7842104a9f86fbf419326b7aad08ade61d
but with this (2024-12-02)) - no: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=6c9bc0156a8d36d56735831cb81e786d628ed73e
Probably somewhere between, is the one which causing problems.

HommeOursPorc, 2025-03-11:
> Images with - this one (2024-10-04) work: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=f006be7842104a9f86fbf419326b7aad08ade61d but with this (2024-12-02)) - no: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=6c9bc0156a8d36d56735831cb81e786d628ed73e Probably somewhere between, is the one which causing problems (~67 commits).

Thanks 😊 

I've sent this on the dnsmasq mailing list. From what I understand this is how I can send a bug report to them.

HommeOursPorc, 2025-03-12:
Voilà: https://www.mail-archive.com/dnsmasq-discuss@lists.thekelleys.org.uk/msg17898.html

pedro0311, 2025-03-12:
Update: dnsmasq crashes also on: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=498794ad85698afb3bcb93aada7559dbf0c177dd
So we have now 16 commits to check.

pedro0311, 2025-03-12:
Update: dnsmasq works on a9d46d4: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=a9d46d42cb2da24c6eb6c887b948fb0c71d4c094

So something between a9d46d4 and (including) 498794a.

pedro0311, 2025-03-12:
959d991 crashes so between (including):
https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=3b6df06fb8cb3652d2e7afd085fae3f416408013
and
https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=ed6d29a78475f9ec91141120aba53490bc1dc39a

pedro0311, 2025-03-12:
Found it: https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=3b6df06fb8cb3652d2e7afd085fae3f416408013
(Always save forwarded query locally.)

A short explanation of why I didn't find this issue earlier (not because I wouldn't check):
- this issue only occurs for non-WAN configuration (not in my case) OR is visible during the first start of dnsmasq, when WAN is still DOWN,
- I use USB logging, which starts later than dnsmasq is started for the first time, so can't see this crash, and when WAN is up - dnsmasq (after the crash) is (re-)started again and everything seems to be fine.

So the question is: how to fix this 3b6df06 commit?

pedro0311, 2025-03-12:
BTW, as far as I can see, this commit can't be simply reverted...

HommeOursPorc, 2025-03-12:
Updated dnsmasq mailing list message: 
https://www.mail-archive.com/dnsmasq-discuss@lists.thekelleys.org.uk/msg17900.html

tisj, 2025-03-12:
The commit you mention is about DNS requests (UDP), when they are being forwarded to upstream DNS servers.

In the syslog.txt I see that there is no mention of upstream servers reported by dnsmasq

> Jan  1 01:00:13 unknown daemon.info dnsmasq[376]: asynchronous logging enabled, queue limit is 5 messages
> Jan  1 01:00:13 unknown daemon.warn dnsmasq[376]: no servers found in /etc/resolv.dnsmasq, will retry
> Jan  1 01:00:13 unknown daemon.info dnsmasq[376]: read /etc/hosts - 3 names

Do you think this could be related?

Can you post dnsmasq configuration?

Is a core file and/or stack trace available?

Did you try to reproduce the issue on a local machine?

HommeOursPorc, 2025-03-12:
> Can you post dnsmasq configuration?

This is on 2024.4, but AFAIK should be the same as 2025.1

```
root@unknown:/etc# cat dnsmasq.conf
pid-file=/var/run/dnsmasq.pid
resolv-file=/etc/resolv.dnsmasq
expand-hosts
min-port=4096
no-negcache
dhcp-name-match=set:wpad-ignore,wpad
dhcp-ignore-names=tag:wpad-ignore
stop-dns-rebind
rebind-localhost-ok
bogus-priv
domain-needed
interface=br0
dhcp-range=tag:br0,192.168.1.2,192.168.1.50,255.255.255.0,1440m
dhcp-option=tag:br0,3,192.168.1.1
addn-hosts=/etc/hosts.dnsmasq
dhcp-lease-max=255
dhcp-option=252,"\n"
dhcp-authoritative
edns-packet-max=1232

root@unknown:/etc# cat hosts.dnsmasq
192.168.1.1 unknown
127.0.0.1 wan1-ip
127.0.0.1 wan2-ip
root@unknown:/etc# cat /etc/resolv.dnsmasq
cat: can't open '/etc/resolv.dnsmasq': No such file or directory
```

@pedro0311 is it normal that `/etc/resolv.dnsmasq` is missing???

pedro0311, 2025-03-12:
> The commit you mention is about DNS requests (UDP), when they are being forwarded to upstream DNS servers.

Right, and something is wrong there, when WAN is down, it can't find it and spits sigsegv.

> @pedro0311 is it normal that /etc/resolv.dnsmasq is missing???
> Do you think this could be related?

This is standard behavior: it will be updated (added) when WAN is up (without this we don't know what the resolver is).

> Just a notice: this is a release candidate, not an official release, so probably not a good idea to include in a (production) release.

Honestly, from my many years of experience I can say that I don't see any differences between beta/test, RC and official release of a given package. Sometimes beta or RC versions are more reliable than official releases. ;)

> Is a core file and/or stack trace available?

This is a problem in our case, but it seems doable. How would you like to invoke dnsmasq? Should I understand that I have to build it with -ggdb and with preserved symbols?

BTW, dnsmasq is built with: ```-DHAVE_BROKEN_RTC -DNO_ID -DNO_GMP -DUSE_IPSET -DHAVE_DNSSEC -DHAVE_DNSSEC_STATIC -DNO_GOST -DUSE_IPV6```

and additionally with  ```-DHAVE_TOMATO``` (our own implementation needed by FreshTomato): https://github.com/FreshTomato-Project/freshtomato-arm/blob/arm-master/release/src-rt-6.x.4708/router/patches/dnsmasq/101-tomato-specific.patch

tisj, 2025-03-12:
I would try to reproduce this on a normal machine, with minimal settings, specifically to highlight the real issue.

Since it is a C program, -g and alikes should be fine. The advantage to reproduce this on a dev machine: you can run it under gdb.

If an empty list of upstream servers is the issue, that should be easily reproducible. HAVE_DNSSEC  COPT could be a valid pointer.

The TOMATO specific patch seems rather harmless, though it is related to DHCP. Maybe you could try to leave it out of your build and retest, just to make sure.

pedro0311, 2025-03-12:
Just to be clear: on both starts of this router (see syslog), I have the same config and the same working conditions.
Removing TOMATO patch doesn't change (help) anything:

```
On commit a9d46d4:

Jan  1 01:00:15 n18u daemon.info dnsmasq[930]: started, version  (origin/tcp-trunc, holly/tcp-trunc, tcp-trunc) cachesize 4096
Jan  1 01:00:15 n18u daemon.info dnsmasq[930]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:15 n18u daemon.info dnsmasq[930]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:15 n18u daemon.info dnsmasq-dhcp[930]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:24 n18u daemon.info dnsmasq-dhcp[930]: DHCPDISCOVER(br0) xx:xx
Jan  1 01:00:24 n18u daemon.info dnsmasq-dhcp[930]: DHCPOFFER(br0) 192.168.1.21 xx:xx
Jan  1 01:00:24 n18u daemon.info dnsmasq-dhcp[930]: DHCPREQUEST(br0) 192.168.1.21 xx:xx
Jan  1 01:00:24 n18u daemon.info dnsmasq-dhcp[930]: DHCPACK(br0) 192.168.1.21 xx:xx laptop
/.../
Jan  1 01:00:28 n18u user.info preinit[1]: Asus RT-N18U: FreshTomato 2025.1.019 K26ARM beta dnsmasq-issue_a9d46d4 USB AIO-64K-NOSMP

and dnsmasq is up and running.

On commit 3b6df06:

Jan  1 01:00:15 n18u daemon.info dnsmasq[865]: started, version  cachesize 4096
Jan  1 01:00:15 n18u daemon.info dnsmasq[865]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:15 n18u daemon.info dnsmasq[865]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:15 n18u daemon.info dnsmasq-dhcp[865]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:15 n18u daemon.info dnsmasq[865]: read /etc/hosts - 3 names
Jan  1 01:00:15 n18u daemon.info dnsmasq[865]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:17 n18u daemon.info httpd[868]: generating SSL certificate...
Jan  1 01:00:22 n18u daemon.info dnsmasq-dhcp[865]: DHCPDISCOVER(br0) xx:xx
Jan  1 01:00:22 n18u daemon.info dnsmasq-dhcp[865]: DHCPOFFER(br0) 192.168.1.21 xx:xx
Jan  1 01:00:22 n18u daemon.info dnsmasq-dhcp[865]: DHCPREQUEST(br0) 192.168.1.21 xx:xx 
Jan  1 01:00:22 n18u daemon.info dnsmasq-dhcp[865]: DHCPACK(br0) 192.168.1.21 xx:xx laptop
Jan  1 01:00:23 n18u kern.warn kernel: dnsmasq/865: potentially unexpected fatal signal 11.
Jan  1 01:00:23 n18u kern.warn kernel: Pid: 865, comm:              dnsmasq
Jan  1 01:00:23 n18u kern.warn kernel: CPU: 0    Tainted: P             (2.6.36.4brcmarm #16)
Jan  1 01:00:23 n18u kern.warn kernel: PC is at 0x2c4b4
Jan  1 01:00:23 n18u kern.warn kernel: LR is at 0x2c474
Jan  1 01:00:23 n18u kern.warn kernel: pc : [<0002c4b4>]    lr : [<0002c474>]    psr: 60000010
Jan  1 01:00:23 n18u kern.warn kernel: sp : be816a38  ip : 000a2851  fp : 000a2828
Jan  1 01:00:23 n18u kern.warn kernel: r10: 00000029  r9 : 00000000  r8 : 000a4fd0
Jan  1 01:00:23 n18u kern.warn kernel: r7 : 00000000  r6 : 0000000e  r5 : 000a1c18  r4 : 00000000
Jan  1 01:00:23 n18u kern.warn kernel: r3 : 000a1c18  r2 : 00000000  r1 : 0000000c  r0 : 00000000
Jan  1 01:00:23 n18u kern.warn kernel: Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Jan  1 01:00:23 n18u kern.warn kernel: Control: 10c53c7d  Table: 9db28059  DAC: 00000015
/.../
Jan  1 01:00:29 n18u user.info preinit[1]: Asus RT-N18U: FreshTomato 2025.1.020 K26ARM beta dnsmasq-issue_3b6df06 USB AIO-64K-NOSMP

and of course it's down.

On commit 3b6df06 without TOMATO patch (no change):

Jan  1 01:00:15 n18u daemon.info dnsmasq[945]: started, version  cachesize 4096
Jan  1 01:00:15 n18u daemon.info dnsmasq[945]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:15 n18u daemon.info dnsmasq[945]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:15 n18u daemon.info dnsmasq-dhcp[945]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:15 n18u daemon.info dnsmasq[945]: read /etc/hosts - 3 names
Jan  1 01:00:15 n18u daemon.info dnsmasq[945]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:24 n18u daemon.info dnsmasq-dhcp[945]: DHCPDISCOVER(br0) xx:xx
Jan  1 01:00:24 n18u daemon.info dnsmasq-dhcp[945]: DHCPOFFER(br0) 192.168.1.21 xx:xx
Jan  1 01:00:24 n18u daemon.info dnsmasq-dhcp[945]: DHCPREQUEST(br0) 192.168.1.21 xx:xx
Jan  1 01:00:24 n18u daemon.info dnsmasq-dhcp[945]: DHCPACK(br0) 192.168.1.21 xx:xx laptop
Jan  1 01:00:24 n18u kern.warn kernel: dnsmasq/945: potentially unexpected fatal signal 11.
Jan  1 01:00:24 n18u kern.warn kernel: Pid: 945, comm:              dnsmasq
Jan  1 01:00:24 n18u kern.warn kernel: CPU: 0    Tainted: P             (2.6.36.4brcmarm #20)
Jan  1 01:00:24 n18u kern.warn kernel: PC is at 0x2c3f8
Jan  1 01:00:24 n18u kern.warn kernel: LR is at 0x2c3b8
Jan  1 01:00:24 n18u kern.warn kernel: pc : [<0002c3f8>]    lr : [<0002c3b8>]    psr: 60000010
Jan  1 01:00:24 n18u kern.warn kernel: sp : beb92a38  ip : 000a2851  fp : 000a2828
Jan  1 01:00:24 n18u kern.warn kernel: r10: 00000029  r9 : 00000000  r8 : 000a4fd0
Jan  1 01:00:24 n18u kern.warn kernel: r7 : 00000000  r6 : 0000000e  r5 : 000a1c18  r4 : 00000000
Jan  1 01:00:24 n18u kern.warn kernel: r3 : 000a1c18  r2 : 00000000  r1 : 0000000c  r0 : 00000000
Jan  1 01:00:24 n18u kern.warn kernel: Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
Jan  1 01:00:24 n18u kern.warn kernel: Control: 10c53c7d  Table: 9da6c059  DAC: 00000015
/.../
Jan  1 01:00:26 n18u user.info preinit[1]: Asus RT-N18U: FreshTomato 2025.1.022 K26ARM beta dnsmasq-issue_3b6df06-no_TOMATO USB AIO-64K-NOSMP
```

I'll try now to build it with debug symbols.
BR

pedro0311, 2025-03-12:
OK, so how should I run dnsmasq under gdb?

pedro0311, 2025-03-12:
Well, I didn't even have to use gdb to find out what is wrong. I added logging (patch here: https://bitbucket.org/pedro311/freshtomato-arm/downloads/forward-log.patch) and now you can see the issue clearly:
```
commit 3b6df06:
(**first** is defined as **int**)

Jan  1 01:00:16 n18u daemon.info dnsmasq[947]: started, version  cachesize 4096
Jan  1 01:00:16 n18u daemon.info dnsmasq[947]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:16 n18u daemon.info dnsmasq[947]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:16 n18u daemon.info dnsmasq-dhcp[947]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:16 n18u daemon.warn dnsmasq[947]: *** fast_retry: IN
Jan  1 01:00:16 n18u daemon.info dnsmasq[947]: read /etc/hosts - 3 names
Jan  1 01:00:16 n18u daemon.info dnsmasq[947]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:23 n18u daemon.info dnsmasq-dhcp[947]: DHCPDISCOVER(br0)
Jan  1 01:00:23 n18u daemon.info dnsmasq-dhcp[947]: DHCPOFFER(br0)
Jan  1 01:00:23 n18u daemon.info dnsmasq-dhcp[947]: DHCPREQUEST(br0)
Jan  1 01:00:23 n18u daemon.info dnsmasq-dhcp[947]: DHCPACK(br0)
Jan  1 01:00:23 n18u daemon.warn dnsmasq[947]: *** receive_query: IN
Jan  1 01:00:23 n18u daemon.warn dnsmasq[947]: *** log_query_mysockaddr: IN
Jan  1 01:00:23 n18u daemon.warn dnsmasq[947]: *** forward_query: IN
Jan  1 01:00:23 n18u daemon.warn dnsmasq[947]: *** lookup_frec_by_query: IN
Jan  1 01:00:23 n18u daemon.warn dnsmasq[947]: *** forward_query: new query
Jan  1 01:00:23 n18u daemon.warn dnsmasq[947]: *** forward_query: if !(lookup_domain
Jan  1 01:00:23 n18u daemon.warn dnsmasq[947]: *** forward_query: before master =
Jan  1 01:00:23 n18u daemon.warn dnsmasq[947]: *** forward_query: **first**=[-1095616084]
Jan  1 01:00:23 n18u kern.warn kernel: dnsmasq/947: potentially unexpected fatal signal 11.
```

It's rather obvious that this can't work, hence sigsegv...

```
commit a9d46d4:

Jan  1 01:00:18 n18u daemon.info dnsmasq[934]: started, version  (origin/tcp-trunc, holly/tcp-trunc, tcp-trunc) cachesize 4096
Jan  1 01:00:18 n18u daemon.info dnsmasq[934]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:18 n18u daemon.info dnsmasq[934]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:18 n18u daemon.info dnsmasq-dhcp[934]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:18 n18u daemon.info dnsmasq[934]: read /etc/hosts - 3 names
Jan  1 01:00:18 n18u daemon.info dnsmasq[934]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:23 n18u daemon.info dnsmasq-dhcp[934]: DHCPDISCOVER(br0)
Jan  1 01:00:23 n18u daemon.info dnsmasq-dhcp[934]: DHCPOFFER(br0)
Jan  1 01:00:23 n18u daemon.info dnsmasq-dhcp[934]: DHCPREQUEST(br0) 
Jan  1 01:00:23 n18u daemon.info dnsmasq-dhcp[934]: DHCPACK(br0)
Jan  1 01:00:24 n18u daemon.warn dnsmasq[934]: *** log_query_mysockaddr: IN
Jan  1 01:00:24 n18u daemon.warn dnsmasq[934]: *** forward_query: IN
Jan  1 01:00:24 n18u daemon.warn dnsmasq[934]: *** forward_query: new query
Jan  1 01:00:24 n18u daemon.warn dnsmasq[934]: *** forward_query: if !(lookup_domain
Jan  1 01:00:24 n18u daemon.warn dnsmasq[934]: *** forward_query: if (flags || ede
Jan  1 01:00:24 n18u daemon.warn dnsmasq[934]: *** send_from: IN
Jan  1 01:00:24 n18u daemon.warn dnsmasq[934]: *** log_query_mysockaddr: IN
/.../
dnsmasq is up
```

As you can see, on last working commit, if we have "new query" it do not even touch ```master = daemon->serverarray[first];```, because **first** in this case has some stupid value.

So IMO, there is a logical problem here that should be solved.
Let me know.

BR

pedro0311, 2025-03-13:
OK, I'll try with latest master and extended logging, and will let you know.

pedro0311, 2025-03-13:
Well, the problem still persists (patch: https://bitbucket.org/pedro311/freshtomato-arm/downloads/forward-master-log.patch):
```
Jan  1 01:00:16 n18u daemon.info dnsmasq[945]: started, version 2.91rc5-e427d4b cachesize 4096
Jan  1 01:00:16 n18u daemon.info dnsmasq[945]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:16 n18u daemon.info dnsmasq[945]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:16 n18u daemon.info dnsmasq-dhcp[945]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:16 n18u daemon.info dnsmasq[945]: read /etc/hosts - 3 names
Jan  1 01:00:16 n18u daemon.info dnsmasq[945]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:19 n18u user.info preinit[1]: Asus RT-N18U: FreshTomato 2025.1.030 K26ARM beta dnsmasq-issue_master-gdb-log USB AIO-64K-NOSMP
Jan  1 01:00:22 n18u daemon.info dnsmasq-dhcp[945]: DHCPDISCOVER(br0)
Jan  1 01:00:22 n18u daemon.info dnsmasq-dhcp[945]: DHCPOFFER(br0)
Jan  1 01:00:22 n18u daemon.info dnsmasq-dhcp[945]: DHCPREQUEST(br0)
Jan  1 01:00:22 n18u daemon.info dnsmasq-dhcp[945]: DHCPACK(br0)
Jan  1 01:00:23 n18u daemon.warn dnsmasq[945]: *** log_query_mysockaddr: IN
Jan  1 01:00:23 n18u daemon.warn dnsmasq[945]: *** forward_query: IN
Jan  1 01:00:23 n18u daemon.warn dnsmasq[945]: *** forward_query: new query
Jan  1 01:00:23 n18u daemon.warn dnsmasq[945]: *** forward_query: if (!lookup_domain()		<----- (!!!)
Jan  1 01:00:23 n18u daemon.warn dnsmasq[945]: *** forward_query: before master =
Jan  1 01:00:23 n18u daemon.warn dnsmasq[945]: *** forward_query: first=[41]
Jan  1 01:00:23 n18u kern.warn kernel: dnsmasq/945: potentially unexpected fatal signal 11.
/.../
Jan  1 01:00:41 n18u user.notice switch4g[966]: 4G MODEM - connected (device /dev/ttyUSB0) ...	<----- WAN up
Jan  1 01:00:43 n18u daemon.info dnsmasq[1386]: started, version 2.91rc5-e427d4b cachesize 4096	<----- started (was down)
Jan  1 01:00:43 n18u daemon.info dnsmasq[1386]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP no-conntrack ipset no-nftset Tomato-helper auth DNSSEC no-ID loop-detect inotify dumpfile
Jan  1 01:00:43 n18u daemon.info dnsmasq[1386]: asynchronous logging enabled, queue limit is 5 messages
Jan  1 01:00:43 n18u daemon.info dnsmasq-dhcp[1386]: DHCP, IP range 192.168.1.2 -- 192.168.1.50, lease time 1d
Jan  1 01:00:43 n18u daemon.info dnsmasq[1386]: reading /etc/resolv.dnsmasq
Jan  1 01:00:43 n18u daemon.info dnsmasq[1386]: using nameserver 89.108.195.21#53
Jan  1 01:00:43 n18u daemon.info dnsmasq[1386]: using nameserver 185.89.185.1#53
Jan  1 01:00:43 n18u daemon.info dnsmasq[1386]: read /etc/hosts - 3 names
Jan  1 01:00:43 n18u daemon.info dnsmasq[1386]: read /etc/hosts.dnsmasq - 5 names
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** log_query_mysockaddr: IN
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** forward_query: IN
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** forward_query: new query
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** forward_query: if (lookup_domain()		<----- (!!!)
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** forward_query: before master =
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** forward_query: first=[0]
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** forward_query: after: master =		<----- all good
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** domain_no_rebind: IN
Jan  1 01:00:43 n18u daemon.warn dnsmasq[1386]: *** log_query_mysockaddr: IN
/.../
```

Now, I'm almost sure, that:
```
      master = daemon->serverarray[first];
```

**must** be moved after:
```
      /* Configured answer. */
      if (flags || ede == EDE_NOT_READY)
	goto reply;
```

to avoid such issues, at least in our case (old kernel + uclibc).

Just checking it.

pedro0311, 2025-03-13:
I was right, no more SIGSEGV, everything working correctly. :)
I have to check also MIPS branch now.

pedro0311, 2025-03-13:
Exactly, so please could you fix this upstream?
Thanks & BR

tisj, 2025-03-13:
We can send in this patch:

```
commit 0d87b0ac8ed525ab1eb43b753145702eba0db197 (HEAD -> master)
Author: Tijs Van Buggenhout <tijs.van.buggenhout@axsguard.com>
Date:   Thu Mar 13 14:42:10 2025 +0100

    Partially revert "Always save  forwarded query locally."
    
    This partially reverts commit 3b6df06fb8cb3652d2e7afd085fae3f416408013.

diff --git a/src/forward.c b/src/forward.c
index 8207a7e..939a4dc 100644
--- a/src/forward.c
+++ b/src/forward.c
@@ -353,9 +353,7 @@ static void forward_query(int udpfd, union mysockaddr *udpaddr,
          ede = EDE_NOT_READY;
          flags = 0;
        }
-
-      master = daemon->serverarray[first];
-
+       
       /* don't forward A or AAAA queries for simple names, except the empty name */
       if (!flags &&
          option_bool(OPT_NODOTS_LOCAL) &&
@@ -368,6 +366,8 @@ static void forward_query(int udpfd, union mysockaddr *udpaddr,
       if (flags || ede == EDE_NOT_READY)
        goto reply;
       
+      master = daemon->serverarray[first];
+      
       if (!(forward = get_new_frec(now, master, 0)))
        goto reply;
       /* table full - flags == 0, return REFUSED */
```

pedro0311, 2025-03-13:
Sure, go on, AFAIK, I'm not subscribed to the @Simon mailing list (or maybe I am? NM).

MIPS branch also works now correctly.

tisj, 2025-03-13:
https://lists.thekelleys.org.uk/pipermail/dnsmasq-discuss/2025q1/018072.html

closed 2025-03-14 by pedro0311

pedro0311, 2025-03-14:
@tisj : since you are already on the mailing list @Simon could you point him to one problem I noticed when adding logging to the forward.c file - this format is accepted by the compiler (no warnings or something):

```my_syslog(LOG_ERR, _("something %s something %s"), __FUNCTION__, __LINE__);```

but it causes an immediate crash during code execution. So (in my opinion) something is wrong here - either this (adding more than one %s) should be prevented at the compilation stage, or my_syslog() function should allow adding more than just one variable.

HommeOursPorc, 2025-03-14:
dnsmasq commit: 
https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=commitdiff;h=9af15871e6ef0ac845412e9a6c3ecfb0849374c8

commit 0fc276d9 (2025-03-14): dnsmasq: fix crash after assigning IP address when WAN is still down (close #6)

- see: https://github.com/FreshTomato-Project/freshtomato-mips/issues/6